Klemen1337/node-thermal-printer #160: printer.isPrinterConnected() giving true even for a wrong IP when used with mobile hotspot

Akshay-Jannu, 2020-12-02:
**let printer = new ThermalPrinter({
      type: PrinterTypes.EPSON,
      interface: "tcp://192.168.1.",
    });

    let isConnected = await printer.isPrinterConnected();**

_isConnected is coming as true even though the IP is wrong._

 **let execute =await printer.execute();**

_execute is coming as Data sent to printer: 192.168.1.:9100_

This is happening when I use a mobile hotspot, is there any solution for this, or may be an alternative?

shabanbx, 2021-01-01:
same problem

afif-bouzid, 2021-04-29:
I have false when I use the print share

          interface: '//localhost/printer',
     
        });
Do you have a solution?

hoangphan84, 2021-05-21:
I have submitted a PR to check the status of the printer and return proper isPrinterConnected value
https://github.com/Klemen1337/node-thermal-printer/pull/171

In the meantime, I suggest to use the printer driver to manually check the status of the printer and determine its connection status.

closed 2022-04-16 by stale[bot]