Klemen1337/node-thermal-printer #161: Error: No driver set!

blank-tree, 2020-12-03:
I'm using macOS, using Node.js v13.14.0 and installed the [printer](https://www.npmjs.com/package/printer) library. 

I'm using this code:

```js
const printer = require('printer');
const ThermalPrinter = require("node-thermal-printer").printer;
const Types = require("node-thermal-printer").types;

async function example () {
	let out = new ThermalPrinter({
		type: Types.EPSON,
		interface: 'printer:EPSON_TM_T20II',
		driver: 'printer',
	    width: 48,
	    characterSet: 'SLOVENIA',
	    removeSpecialCharacters: false,
	    lineCharacter: "-",
	});

	let isConnected = await out.isPrinterConnected();
	console.log("Printer connected:", isConnected);

	out.alignLeft();
	out.newLine();
	out.println("Hello World!");
	out.drawLine();

	out.cut();

	try {
		await out.execute();
		console.log("Print success.");
	} catch (error) {
		console.error("Print error:", error);
	}
}

example();
```

But it always terminates with:

```bash
(node:27391) UnhandledPromiseRejectionWarning: Error: No driver set!
```

What am I doing wrong?

The printer package is returnig with:
```js
const printer = require('printer');
console.log(printer.getPrinter('EPSON_TM_T20II'));
```

the following:

```json
{
  name: 'EPSON_TM_T20II',
  isDefault: false,
  options: {
    copies: '1',
    'device-uri': 'usb://EPSON/TM-T20II?serial=544338595827330000',
    finishings: '3',
    'job-cancel-after': '10800',
    'job-hold-until': 'no-hold',
    'job-priority': '50',
    'job-sheets': 'none,none',
    'marker-change-time': 1970-01-01T00:00:00.000Z,
    'number-up': '1',
    'printer-commands': 'none',
    'printer-info': 'EPSON TM-T20II',
    'printer-is-accepting-jobs': 'true',
    'printer-is-shared': 'false',
    'printer-is-temporary': 'false',
    'printer-location': 'HAL',
    'printer-make-and-model': 'EPSON TM-T20II (rastertotmt)',
    'printer-state': '3',
    'printer-state-change-time': 2020-12-03T15:12:51.000Z,
    'printer-state-reasons': 'none',
    'printer-type': '2134020',
    'printer-uri-supported': 'ipp://localhost/printers/EPSON_TM_T20II'
  },
  status: 'IDLE'
}
```

Edig, 2020-12-10:
I have the same problem with the same printer, any ideas?

blank-tree, 2020-12-10:
I've found a solution. I'm not sure where the problem is, but this way it worked out.

Software:
* macOS 10.15.7
* node.js 13.14.0
* npm 6.14.4
* nvm 0.37.2
* CUPS

NPM:
* [node-printer](https://www.npmjs.com/package/printer)
* [node-thermal-printer](https://www.npmjs.com/package/node-thermal-printer)

Setup CUPS:
* Install EPSON driver see https://download.epson-biz.com/modules/pos/index.php?page=single_soft&cid=6407&scat=33&pcat=3
* activate CUPS web interface through the commandline ```cupsctl WebInterface=yes```
* connect to through a browser [http://localhost:631](http://localhost:631)
* setup printer from Administration
* note the exact name (without spaces) for the application
* Make sure to disable cupctl web interface when done ```cupsctl WebInterface=no```

Code:
```js
const printer = require('printer');
const ThermalPrinter = require("node-thermal-printer").printer;
const Types = require("node-thermal-printer").types;

async function printMsg(msg) {
    let epsonThermalPrinter = new ThermalPrinter({
        type: Types.EPSON,
        width: 48,
        characterSet: 'SLOVENIA',
        removeSpecialCharacters: false,
        lineCharacter: "-",
    });

    epsonThermalPrinter.newLine();
    epsonThermalPrinter.println(msg);
    epsonThermalPrinter.newLine();

    epsonThermalPrinter.printDirect({
        data: epsonThermalPrinter.getBuffer(),
        printer: 'EPSON_TM-T20II',
        type: 'RAW',
        success: function (jobID) {
            console.log(`printer job: ${jobID}`);
            epsonThermalPrinter.clear();
        },
        error: function (err) {
            console.log(err);
        }
    })
}

printMsg('foobar');
```

closed 2020-12-10 by blank-tree

SwapnilSoni1999, 2021-03-07:
> I've found a solution. I'm not sure where the problem is, but this way it worked out. @Edig
> 
> Software:
> 
> * macOS 10.15.7
> * node.js 13.14.0
> * npm 6.14.4
> * nvm 0.37.2
> * CUPS
> 
> NPM:
> 
> * [node-printer](https://www.npmjs.com/package/printer)
> * [node-thermal-printer](https://www.npmjs.com/package/node-thermal-printer)
> 
> Setup CUPS (I got this from this [issue](https://github.com/Klemen1337/node-thermal-printer/issues/86)):
> 
> * Install EPSON driver see https://download.epson-biz.com/modules/pos/index.php?page=single_soft&cid=6407&scat=33&pcat=3
> * activate CUPS web interface through the commandline `cupsctl WebInterface=yes`
> * connect to through a browser http://localhost:631
> * setup printer from Administration
> * note the exact name (without spaces) for the application
> * Make sure to disable cupctl web interface when done `cupsctl WebInterface=no`
> 
> Code:
> 
> ```js
> const printer = require('printer');
> const ThermalPrinter = require("node-thermal-printer").printer;
> const Types = require("node-thermal-printer").types;
> 
> async function printMsg(msg) {
>     let epsonThermalPrinter = new ThermalPrinter({
>         type: Types.EPSON,
>         width: 48,
>         characterSet: 'SLOVENIA',
>         removeSpecialCharacters: false,
>         lineCharacter: "-",
>     });
> 
>     epsonThermalPrinter.newLine();
>     epsonThermalPrinter.println(msg);
>     epsonThermalPrinter.newLine();
> 
>     printer.printDirect({
>         data: epsonThermalPrinter.getBuffer(),
>         printer: 'EPSON_TM-T20II',
>         type: 'RAW',
>         success: function (jobID) {
>             console.log(`printer job: ${jobID}`);
>             epsonThermalPrinter.clear();
>         },
>         error: function (err) {
>             console.log(err);
>         }
>     })
> }
> 
> printMsg('foobar');
> ```

Thanks alot it took me half of the day to figure out either node-printer or electron-printer is required

thiagosanches, 2023-06-21:
Hey guys, you can check this:
https://github.com/thiagoelg/node-printer