Klemen1337/node-thermal-printer #164: How can I add another language charset support?

denven, 2021-01-19:
I found that I cannot print Chinese-Simplified Characters when I set it "CHINA" when printing on a EPSON TM-T88VI printer, is there any way to support this charsets?

Thanks.

denven, 2021-01-21:
> I cannot print Chinese-Simplified Characters on a EPSON TM-T88VI printer via tcp printing even when I set the `characterSet` to 'CHINA', I checked the ESC charsets commands from [EPSON doc](https://reference.epson-biz.com/modules/ref_escpos/index.php?content_id=29), it's already added in `epson-config.js` file:
> 
> ```js
> CODE_PAGE_CHINA                 : Buffer.from([0x1b, 0x52, 0x0F]),
> ```
> 
> But when I initite the printer with:
> ```js
> characterSet: 'CHINA',
> ```
> 
> it doesn't work.
> 
> And I also find that the example code `charset.js` which sets the Chinese charset cannot print Chinese, Japanese and Hebrew.
> ```js
>     printer.setCharacterSet('PC855_CYRILLIC');
>     printer.println("Все люди рождаются свободными и равными в своем достоинстве и правах.");
>     printer.drawLine();
>     printer.setCharacterSet('JAPAN');
>     printer.println("すべての人々は自由に生まれ、尊厳と権利において平等です");
>     printer.drawLine();
>     printer.setCharacterSet('CHINA');
>     printer.println("所有人生而自由，在尊嚴和權利上一律平等");
>     printer.drawLine();
>     printer.setCharacterSet('PC862_HEBREW');
>     printer.println("כל בני האדם נולדים חופשיים ושווים בכבוד ובזכויות.");
> ```
> 
> Where is the issue? 
> 
> Thanks.

I found the reason! 

Epson printer has different Models, some type has several  models under the same name. And different model may vary in languages char set support.

The model will output garbled code for some language given that you've set language char set, that is due to code page for that language is not included for the model, and there is no corresponding font in the firmware to print right.

To fix, contact the seller or EPSON to update/add font support. You may need pay some extra fee.

hawkerboy7, 2021-03-03:
The solution I am using is to print images of the characters instead.
This way you can show characters even though the printer does not have/support the languages char set.
This works if you have a manageable fixed amount of Chinese, Japanese or Hebrew charters for example in a restaurant printing the dishes for the chefs.

closed 2021-07-21 by stale[bot]

TraFost, 2024-04-17:
> The solution I am using is to print images of the characters instead. This way you can show characters even though the printer does not have/support the languages char set. This works if you have a [...]

how do u implement this?

hawkerboy7, 2024-04-17:
I've copy pasted most of the pieces of code I used to get a restaurant order printed, without the queing logic.
I wrote this in coffeescript back then though..

Two "problems":
1. A failure to connect however does not trigger the catch, I could not find what went wrong there.
2. The printers I had, had a limited buffer. Sending to many orders would cause the last sent orders to be ignored. The library and/or printer would however not throw an error anywhere, so you won't notice this easily. I resolved it by measuring how long each order takes to print (between 500ms/1 second) and then make the queue logic wait that maximum time before sending the next item to print.    

```coffeescript
NTP = require "node-thermal-printer"

printer = new NTP.printer
	type         : NTP.types.EPSON
	width        : 24,
	interface    : "tcp://#{record.ip}"
	characterSet : "SLOVENIA"

printer.isPrinterConnected()
	.then =>

		@log.info "Printer [name] connected"

		printAnOrder()

	.catch (e) =>

		# This never runs though

printAnOrder = ->

	# Printer default size + header
	printer.setTextSize 1, 1

	# Take the first order as the sample for the table layout
	printer.tableCustom [
		text  : "[table-number]"
		align : "LEFT"
		bold  : true
	,
		text  : "[current-time]"
		align : "CENTER"
	,
		text  : "[table-number]"
		align : "RIGHT"
		bold  : true
	]

	# Draw seperation line
	printer.drawLine()

	printer.print "[normal text about the order]"

	# Add the asian character image
	printer.printImage "[absolute path to image].png"
		.then =>
			# Do something else
		.catch (e) =>
			@log.error "Cannot find character image. Code '#{e.code}', path: '#{e.path}'"
```

TraFost, 2024-04-18:
> I've copy pasted most of the pieces of code I used to get a restaurant order printed, without the queing logic. I wrote this in coffeescript back then though..
> 
> Two "problems":
> 
> 1. A failure to connect however does not trigger the catch, I could not find what went wrong there.
> 2. The printers I had, had a limited buffer. Sending to many orders would cause the last sent orders to be ignored. The library and/or printer would however not throw an error anywhere, so you won't notice this easily. I resolved it by measuring how long each order takes to print (between 500ms/1 second) and then make the queue logic wait that maximum time before sending the next item to print.
> 
> ```coffee
> NTP = require "node-thermal-printer"
> 
> 
> printer = new NTP.printer
> 	type         : NTP.types.EPSON
> 	width        : 24,
> 	interface    : "tcp://#{record.ip}"
> 	characterSet : "SLOVENIA"
> 
> printer.isPrinterConnected()
> 	.then =>
> 
> 		@log.info "Printer [name] connected"
> 
> 		printAnOrder()
> 
> 	.catch (e) =>
> 
> 		# This never runs though
> 
> 
> printAnOrder = ->
> 
> 	# Printer default size + header
> 	printer.setTextSize 1, 1
> 
> 	# Take the first order as the sample for the table layout
> 	printer.tableCustom [
> 		text  : "[table-number]"
> 		align : "LEFT"
> 		bold  : true
> 	,
> 		text  : "[current-time]"
> 		align : "CENTER"
> 	,
> 		text  : "[table-number]"
> 		align : "RIGHT"
> 		bold  : true
> 	]
> 
> 	# Draw seperation line
> 	printer.drawLine()
> 
> 	printer.print "[normal text about the order]"
> 
> 	# Add the asian character image
> 	printer.printImage "[absolute path to image].png"
> 		.then =>
> 			# Do something else
> 		.catch (e) =>
> 			@log.error "Cannot find character image. Code '#{e.code}', path: '#{e.path}'"
> ```

do u have the picture of the printed paper?

hawkerboy7, 2024-04-18:
I could only find this one. Here you see several prints, most have the characters as images, some do not.
[image]

TraFost, 2024-04-18:
but how u be able to convert the value into an image? i'm very confused

hawkerboy7, 2024-04-22:
> but how u be able to convert the value into an image? i'm very confused

Are you asking how to make an image of a character?

Open word/google docs, write the character, increase the font-size, print-screen/screenshot, crop and save as .jpg or .png).
You can link an (order)ID to an image and thus print that image when the ID is requested.

This however is quite some manual work when you have a lot of characters, therefore I've stated before:
> This works if you have a **manageable fixed amount** of Chinese, Japanese or Hebrew charters for example in a restaurant printing the dishes for the chefs.