Klemen1337/node-thermal-printer #168: this.printer.setTextSize is not there for Star Printer

svang-app, 2021-03-03:
If I try to use STAR printer test, it throws an exception on setTextSize.

```
(node:21205) UnhandledPromiseRejectionWarning: TypeError: this.printer.setTextSize is not a function
    at ThermalPrinter.setTextSize (/node-thermal-printer/lib/core.js:204:30)
    at example (/node-thermal-printer/examples/example.js:63:28)
```
Any pointer - how to go about this? Thanks

hawkerboy7, 2021-03-05:
Looking at the code for [star](https://github.com/Klemen1337/node-thermal-printer/blob/master/lib/types/epson.js#L25) that function is not implemented, while it is for [epson](https://github.com/Klemen1337/node-thermal-printer/blob/master/lib/types/epson.js).
Assuming this is left out on purpose a star printer does not support this function.

sidhdharth-navsoft, 2021-03-08:
@hawkerboy7  is this work npm module also work for windows ??
because i am getting printer "ThermalPrinter-- Print done"  for "log.info('ThermalPrinter--', execute);" but it's not printing anything and opening cash drawer for windows pc.

closed 2021-07-21 by stale[bot]