Klemen1337/node-thermal-printer #170: unable to connect to printer

manoj-nexgdata, 2021-03-10:
Im using POS application to print  biill in thermal printer of HOP-E801 printer using node .js ,I have set this as a default printer to my machine.

printer details :
Model :  HOP-E801
width: 58mm-80mm(3 inch bill)
printer name :POS-80-Series

i have installed dependency for printer activity like printer list and print content

Here is my sample test program to connect with printer 

const ThermalPrinter = require("node-thermal-printer").printer;
const PrinterTypes = require("node-thermal-printer").types;

let printer = new ThermalPrinter({
type: PrinterTypes.EPSON,
interface: 'tcp://127.0.0.1:631',

});
myFunction();
async function myFunction() {
let isConnected = await printer.isPrinterConnected();
console.log(isConnected);
printer.println("Check");
printer.cut();
let execute = await printer.execute();
let raw = await printer.raw(Buffer.from("Hello world"));
console.log(raw);
}

Above sample i have tried to connection with printer by port number (127.0.0.1:631) in console it was printing data sent to  127.0.0.1:631

but printer doesn’t get any data 

For the above sample i have changed the interface to printer name :POS-80-Series ,tcp://127.0.0.1:631,i have used all possible ways to connect with printer like socket and network and default ,but no use

manoj-nexgdata, 2021-03-20:
We got blocked in the middle of the development. If some one can guide us to integrate the printer, it will be helpful. I request you people to give a quick response on this.

Thanks.

mitchelson, 2021-04-08:
hi, have you tried `interface: '\\\/localhost\\/PRINTER-NAME',`

afif-bouzid, 2021-04-23:
I advise you to configure your printer in share mode (ex: printer1)
and configure your file like this (
ex: let printer = new ThermalPrinter ({
type: PrinterTypes.EPSON,
interface: '// localhost/printer1',
});
)

closed 2022-04-16 by stale[bot]