Klemen1337/node-thermal-printer #186: driver

WebScaffolder, 2022-02-24:


jeremy-hoot, 2022-02-24:
Hey man I legit just got this error earlier. I found a work around for this. I needed to install this [package](https://www.npmjs.com/package/@thiagoelg/node-printer) and it was actually running. Then i was able to add the driver using this syntax.
`
let printMachine = new ThermalPrinter({
  type: PrinterTypes.STAR,
  interface: "printer:Star TSP143IIIBI Cutter",
  **driver: require("@thiagoelg/node-printer"),**
});`

and it worked for me. I hope this helps!

Mayank-J0shi, 2023-05-09:
It does not work now.. I have tried it multiple times.
```
    const printer = new ThermalPrinter({
      type: PrinterTypes.STAR,
      interface: itfc,
      characterSet: CharacterSet.SLOVENIA,
      removeSpecialCharacters: false,
      lineCharacter: '=',
      breakLine: BreakLine.WORD,
      driver: require('@thiagoelg/node-printer')
    });
```
Gives out error : 
[image]

closed 2023-09-13 by Klemen1337