Klemen1337/node-thermal-printer #192: Epson function for CODE128 actually missing

hesa2020, 2022-03-31:
Hello the function is actually missing. here is how I implemented it for mine:

`this.buffer = null;
this.append(this.config.BARCODE_CODE128);
this.append(Buffer.from([data.length + 2]));
this.append(Buffer.from([0x7b, 0x42]));
this.append(Buffer.from(data.toString()));
return this.buffer;`

`BARCODE_CODE128: Buffer.from([0x1d, 0x6b, 0x49])`

It doesnt use any settings and has no text above/underneat and I am trying to figure how to implement those, however it seems to be working fine.

hesa2020, 2022-03-31:
Using printBarcode 73 didnt worked, it wrote the underneat text and took the space on paper but it wasnt printing any bars.

treeindark, 2022-04-04:
hello, 
I'll use an epson compatible printer, but for code128 i need send :
·When using the CODE 128 in this printer, take the following points into
account for data transmission:
① The top of the bar code data string must be code set selection
character (any of CODE A, CODE B or CODE C) which selects the first code set.
② Special characters are defined by combining two characters
“{“ and one character. The ASCII character “{“ is defined by transmitting “{“ twice
consecutively.
Specific character Transmit data
                  ASCII Hex Decimal
SHIFT      |  {S   |  7B,53  |  123,83
CODE  A  |  {A  |  7B,41  |  123,65
CODE  B  |  {B  |   7B,42  |  123,66
CODE C   |  {C  |  7B,43  |  123,67

I don'know how todo that ... (actualy my code128 is a sadly  UPC_E)

Klemen1337, 2022-10-20:
@hesa2020  Can you make a PR?

racoononmoon, 2022-12-20:
you still can print Code128 AUTO using type 79 instead of 73 without any code adaptations
[https://reference.epson-biz.com/modules/ref_escpos/index.php?content_id=128]

`printer.printBarcode("SOME-CODE128-DATA", 79, {hriPos:4, hriFont:0, width: 2, height: 40}); `

also, same approach as above can be added to the printBarcode function in lib\types\epson.js
the encoding should ideally be a settings parameter
 
```
// Print Barcode
    this.append(Buffer.from([0x1d, 0x6b])); // GS k
    // Select type and bit length of data
    if(type && type == 73){
      this.append(Buffer.from([type, data.length + 2]));
      this.append(Buffer.from([0x7b, 0x41]));   //0x41 -CODE A, 0x42 -CODE B,  0x43 -CODE C, 
    } else{
      this.append(Buffer.from([type, data.length]));
    }
```

TheWhiteShade, 2022-12-23:
> Using printBarcode 73 didnt worked, it wrote the underneat text and took the space on paper but it wasnt printing any bars.
> 
> until i changed:
> 
> ```
> this.append(Buffer.from([type, data.length]));
> ```
> 
> to:
> 
> ```
> if(type == 73)
> {
> 	this.append(Buffer.from([type, data.length + 2]));
> 	this.append(Buffer.from([0x7b, 0x42]));
> }
> else
> {
> 	this.append(Buffer.from([type, data.length]));
> }
> ```
> 
> Note: Tested and working with type 73 and 69.

This worked for me, thank you.

closed 2024-01-23 by Klemen1337