Klemen1337/node-thermal-printer #197: Constructor not set. error

zdridox, 2022-05-16:
HI im checking connection to my star tsp700II with this code and i get an error 'constructor not set'
code:
**```
const ThermalPrinter = require('../node-thermal-printer').printer;
const Types = require('../node-thermal-printer').types;

async function testConnection() {
  const printer = new ThermalPrinter({
    type: Types.STAR,
    interface: 'printer:star', // star is my printers name
    driver: require("@thiagoelg/node-printer"),
  });

  const isConnected = await printer.isPrinterConnected();
  console.log('Printer connected:', isConnected);
}

testConnection();
```**

error:

```
Consturctor not set
Printer connected: true
```

closed 2022-05-17 by zdridox

harish-bytize, 2023-02-16:
@zdridox How did you solve this error?

harish0507, 2023-02-16:
@zdridox How did you solve this error?

pmcp, 2023-06-26:
I have the same issue, can somebody explain what is going on here?