Klemen1337/node-thermal-printer #208: Allow printImage to be centered

xjuanc, 2022-07-25:
Hello everyone, I'm trying to find a way to center an image that is being sent to the printer with the printImage function, I've tried to align it as if it was a text but nothing changed. Is there any way to accomplish this? 

The image prints quite sharp and everything is okay but quite of leaning towards the left side, therefore I'd like to center it. Thanks in advance.

Klemen1337, 2022-10-20:
Is the content of the image in the center?

racoononmoon, 2022-12-20:
are you doing it like this? works for me on a EPSON TM-T88VII and older model EPSON TM-T88V

```
let logoPath = "./logo.png";

printer.alignCenter();

  try {
    await printer.printImage(logoPath);
  } catch (e) {
    console.log(e);
    console.log("printing image failed: " + logoPath);
  }
  printer.cut();
  
   try{
    success = await printer.execute();
    console.log(success);
  } catch(e){
    console.log(e);
  }
```

closed 2023-09-13 by Klemen1337