Klemen1337/node-thermal-printer #209: Reliance on electron-printer/node-printer

mattkrins, 2022-07-26:
This library works brilliantly for printing to networked printers, however I would like to print to a local printer too, but due to the reliance on electron-printer/node-printer this is not possible as these libraries are dead/broken and pull requests are ignored.
Is it possible to print locally without electron-printer/node-printer?

mattkrins, 2022-07-26:
Thanks for the advice @Wolyo, .
I also found a workaround which is going to suit better for me:
On the host running node-thermal-printer share the local printer like you would to the network, but only set permissions for the local user, then for the ThermalPrinter interface just use '//localhost/LocalPrinterName'. No extra library required this way, but it would still be nice to not need workarounds.

SimantoR, 2022-08-15:
Maybe we could build a new C++ Node.JS library for local printers across the 3 platforms. Printer are ancient at this point and finding reference code wouldn't be too hard. I am looking to do things over the network primarily but testing has been a pain since I use a wired one. `thiagoelg/node-printer` seems to be fine, but a more maintained one would be nice.

ojcgv21, 2025-01-03:
@mattkrins An awesome workaround! Thank you!! It works!