Klemen1337/node-thermal-printer #211: GetStatus not implemented

Feeh03114, 2022-07-27:
It does not generate an error when the paper runs out, it would be interesting to get the status if you have paper, give an error that paper is out.

the only thing I can return is the printer data but even that doesn't help because I follow this (https://github.com/tojocky/node-printer/issues/166) tutorial but it doesn't even have the necessary parameters.
 
`{
  name: 'EPSON TM-T20X',
  portName: 'TMUSB001',
  driverName: 'EPSON TM-T(203dpi) Receipt6',
  printProcessor: 'winprint',
  datatype: 'RAW',
  status: [],
  statusNumber: 0,
  attributes: [ 'DO-COMPLETE-FIRST', 'LOCAL', 'RAW-ONLY' ],
  priority: 1,
  defaultPriority: 0,
  averagePPM: 0,
  jobs: [
    {
      id: 28,
      name: 'EPSON TM-T20X',
      printerName: 'EPSON TM-T20X',
      user: 'Totem Homo',
      format: 'RAW',
      priority: 1,
      size: 929,
      status: [],
      machineName: '\\\\DESKTOP-RS415OU',
      document: 'node print job',
      notifyName: 'Totem Homo',
      printProcessor: 'winprint',
      driverName: 'EPSON TM-T(203dpi) Receipt6',
      position: 1,
      startTime: 0,
      untilTime: 0,
      totalPages: 1,
      time: 0,
      pagesPrinted: 0
    }
  ]
}`

racoononmoon, 2022-12-19:
Following the EPSON Command Reference, the printer simply goes OFFLINE when paper is not present.

TM-T88VII:
Paper sensor status (n = 1, 49)
When the roll paper end sensor detects a paper-end, the printer goes offline and does not execute this command. Therefore, bits 2 and 3 of the paper sensor status do not transmit a paper-end status.
https://reference.epson-biz.com/modules/ref_escpos/index.php?content_id=124

you can implement different status queries checking the paper low status and cash drawer status separately.
the difference to the execute() function is that you don't destroy the socket connection immediately after sending data but first  wait for some status bytes to receive from the printer.

call it in the app code like 
`let printerStatus = await printer.status();  `

e.g. in core.js add:

```
async status() {
    let paper = await this.Interface.status("PAPER");
    let drawer = await this.Interface.status("DRAWER");
    let prst = await this.Interface.status("PRINTER");
    return {paper, drawer, printer: prst};
  }
```

in network.js add:
```
async status(which) {

    let buffer;

    if (which == "PAPER"){
      buffer = Buffer.from([0x1D, 0x72, 0x01]);  //03 = paper low, 00 = paper OK
    }else if(which = "DRAWER") {
      buffer = Buffer.from([0x1D, 0x72, 0x02]); //00 = drawer open, 01 = drawer closed
    }else if(which = "PRINTER") {
      buffer = Buffer.from([0x10, 0x04, 0x02]);  //00 = drawer open, 01 = drawer closed
    }
    

    return new Promise((resolve, reject) => {
      let name = this.host + ":" + this.port;
      var printer = Net.connect(
        {
          host: this.host,
          port: this.port,
          timeout: 3000
        },
        function () {
          printer.write(buffer, null, function () {
          });
        }
      );
  
      printer.on('data', function (data) {
        resolve(data.toString("hex"));
        printer.destroy();
      });

      printer.on('error', function (error) {
        resolve("ERROR");
        printer.destroy();
      });
  
      printer.on('timeout', function () {
        resolve("OFFLINE");
        printer.destroy();
      });
    });
  }
```

Klemen1337, 2023-09-13:
I implemented getStatus with the new v4.3.0 release

closed 2023-09-13 by Klemen1337