Klemen1337/node-thermal-printer #212: printImage over TCP/IP results in garbage

geoff8888, 2022-12-09:
Not sure if this is buffer overrun problem, but running this over TCP/IP interface

```
const printer = new ThermalPrinter({
                type: PrinterTypes.EPSON,                                  // Printer type: 'star' or 'epson'
                interface: "tcp://192.168.5.1",                                          // Printer interface
                characterSet: 'SLOVENIA',                                 // Printer character set - default: SLOVENIA
                removeSpecialCharacters: false,                           // Removes special characters - default: false
                lineCharacter: "=",                                       // Set character for lines - default: "-"
                driver: require('printer'),
                options: {                                                 // Additional options
                    timeout: 5000,                                           // Connection timeout (ms) [applicable only for network printers] - default: 3000
                },
            });
await printer.alignCenter();
await printer.printImage(tempFile);
await printer.alignLeft();            
await printer.execute();                   

```

Will result in garbage.

racoononmoon, 2022-12-20:
1) you can remove the line `driver: require('printer')`, you don't need it when printing over tcp
2) save the image in PNG format, use 4bit depth and make it black&white high contrast
3) the size should not be too large, I use about 350x150px and it just works fine

nipoonp, 2023-01-10:
For me, the images print really slowly.

closed 2024-01-23 by Klemen1337