Klemen1337/node-thermal-printer #214: Driver for electron application

atulatulatul, 2023-02-19:
## What's the issue?

Hi. I just wanted to know what should be set as the `driver`.

I just wanted to open the cash drawer attached to the printer `XP-80C` (almost similar to `EPSON`), I've already tried the following packages:

1. `@flovy/node-printer`
2. `@thiagoelg/node-printer`

When using the above packages as driver whenever I execute this line of code:

```
printer.openCashDrawer();
```

The entire electron app crashes and goes blank.

https://user-images.githubusercontent.com/13715884/219932018-616f5ba0-57c6-45b6-af6f-d9977bda6071.mp4

mwsoofy, 2023-03-09:
have you found any solutions or alternatives ?

Mayank-J0shi, 2023-05-09:
I have also been working on this issue although I do not have any printer but using (Microsoft PDF to print) but I am not able to understand what should be the value of driver.
```
    const printer = new ThermalPrinter({
      type: PrinterTypes.STAR,
      interface: itfc,
      characterSet: CharacterSet.SLOVENIA,
      removeSpecialCharacters: false,
      lineCharacter: '=',
      breakLine: BreakLine.WORD,
      driver: require('@thiagoelg/node-printer')
    });
```

This gives out error : 
[image]

adnanlah, 2024-03-30:
> have you found any solutions or alternatives ?

This is one of the alternatives that worked for me. My original plan is to run `node-thermal-printer` without `@thiagoelg/node-printer` because the latter didn't get installed in my electron app.

```ts
let tp = new ThermalPrinter({
    type: PrinterTypes.EPSON, // Printer type: 'star' or 'epson'
    interface: "printer:Xprinter XP-T371U", // Printer interface
    characterSet: CharacterSet.ISO8859_2_LATIN2, // Printer character set
    options: {
      timeout: 5000, // Connection timeout (ms) [applicable only for network printers] - default: 3000
    },
    driver: {}, // Empty object
  });

tp.println("Hello World"); 
tp.table(["One", "Two", "Three"])
tp.cut();

const raw = tp.getBuffer()

let device = usb.findByIds(/*vid*/8137, /*pid*/8214);
if (device) {
    device.open();
    console.log(device.interfaces?.at(0)?.endpoints)
    device.interfaces?.at(0)?.claim();
    const outEndpoint = device.interfaces?.at(0)?.endpoints.find(e => e.direction === 'out');
    if (outEndpoint) {
        outEndpoint.transferType = 2;
        outEndpoint.transfer(raw, (err) => {
            device.close();
        });
    }
}
```

ssxv, 2024-09-09:
https://www.npmjs.com/package/@thiagoelg/node-printer
I have used this package from an electron app.

driver: require(@thiagoelg/node-printer);

closed 2025-05-28 by Klemen1337