Klemen1337/node-thermal-printer #219: Python

nipoonp, 2023-03-23:
Hi guys,

I keep on getting this following error when I'm trying to compile my npm electron project.

I feel theres some error with node-gyp and python.

**"find Python using Python version 3.9.6 found at "/Applications/Xcode.app/Contents/Developer/usr/bin/python3"**

[image]

Can you guys please help me with this. How can I fix this issue?

Do I need to try to install a different version of python or something? My current python version installed is v3.9.6.

rishusam, 2023-03-26:
It seems that your project requires node-gyp and Python 3.9.6, but it is unable to locate Python using the default search paths. You can try the following solutions:

Install Python 3.9.6: Make sure that Python 3.9.6 is installed on your system, and that it is added to your PATH environment variable. You can download Python from the official website: https://www.python.org/downloads/release/python-396/

Specify the path to Python: If Python 3.9.6 is already installed on your system, you can specify the path to it using the --python switch when running npm install or node-gyp. For example:

npm install --python=/path/to/python3.9.6

Update node-gyp: Try updating node-gyp to the latest version by running:

npm install -g node-gyp

Use a Python version manager: If you have multiple versions of Python installed on your system, you can use a Python version manager such as pyenv to manage them. pyenv allows you to easily switch between different versions of Python and ensure that the correct version is used for your project.
Once you have resolved the issue with Python, try running npm install again and see if the error persists.

Klemen1337, 2023-03-27:
This is not a problem of the node-thermal-printer library but node-printer. node-printer needs to be rebuilt for elctron.

closed 2023-03-27 by Klemen1337