Klemen1337/node-thermal-printer #223: Can this be integrated with USB connected thermal printer of any brand

jafsparrow, 2023-05-06:
I have a thermal printer made by the company posway which works with windows installed pos billing software..

my question is , since it not the eposn,star or tanca printer, is there any way to connect to any locally usb connected printers with proper drivers installed?

I am trying to create a webserver with thermal printing capability on server connected printer.

any suggestions? or if there is any other library for the particular purpose plz ...?~!!!!

PedroBarbosaSw, 2023-07-01:
I'm having the same problem. Did you find any solution for that?

jafsparrow, 2023-07-01:
Hi Pedro,
I have selected puppeteer and pdf printer for my server. I still don't know if it is the optimal and efficient solution, but it works  at the moment.

This is how it works now.
1. generated pdf and save.
2. Print pdf and delete.

unaisshemim, 2023-10-05:
HI,
you can use **electron pos printer** for any printer it is connected to the computer.

jafsparrow, 2023-10-12:
> HI, you can use **electron pos printer** for any printer it is connected to the computer.

does electron package work with node js express environment.  I am using nest js as my backend. Need to print to usb or network thermal printer from server... Still no idea how to do it fastly

unaisshemim, 2023-10-26:
No you can't becuase the browzer can't connect directly to the pos printer inorder to achieve this you need electron js to connect.But you can  connect and print using **printjs library** which we can select the printer and print it.

closed 2025-05-28 by Klemen1337