Klemen1337/node-thermal-printer #235: STAR TSP100

jayel74, 2023-09-27:
Hey,

Great library, but we cannot seem to print anything to a TSP 100 STAR printer.
Is does say the data is sended to the printer, but nothing is printed.

Any pointers?

greets
John

Klemen1337, 2023-09-29:
Hi @jayel74 try adding port to your netowrk string. For example "tcp://192.168.18.118:9100"

jayel74, 2023-10-02:
Hey @Klemen1337, i've added 9100, but no go.
Nothing is printed !
#
```
PS D:\development\node\testposprinter> node index.js                                                                  
Print done!
Data sent to printer: 192.168.18.118:9100 <Buffer 1b 74 12 1b 1d 61 01 48 65 6c 6c 6f 20 77 6f 72 6c 64 0a 0b 0b 1b 64 02 1b 40>
PS D:\development\node\testposprinter>

``` 
```javascript
async function testConnection() {
    const printer = new ThermalPrinter({
        type: PrinterTypes.STAR,
        interface: "tcp://192.168.18.118:9100",
    });

    printer.alignCenter();
    printer.println("Hello world");
    //await printer.printImage('./assets/olaii-logo-black.png')
    printer.cut();

    try {
        let execute = printer.execute()
        console.log("Print done!");
    } catch (error) {
        console.error("Print failed:", error);
    }
}
```

Klemen1337, 2023-10-02:
Can you add `await` to execute function? Can you also print the printer config? power on + feed button should print default config. You can check IP and port there.

```node
let execute = await printer.execute()
```

jayel74, 2023-10-02:
```javascript
async function testConnection() {
    const printer = new ThermalPrinter({
        type: PrinterTypes.STAR,
        interface: "tcp://192.168.18.118:9100",
    });

    printer.alignCenter();
    printer.println("Hello world");
    //await printer.printImage('./assets/olaii-logo-black.png')
    printer.cut();

    try {
        let execute = await printer.execute()
        console.log("Print done!");
    } catch (error) {
        console.error("Print failed:", error);
    }
}

testConnection();
``` 

```
PS D:\development\node\testposprinter> node index.js
Data sent to printer: 192.168.18.118:9100 <Buffer 1b 74 12 1b 1d 61 01 48 65 6c 6c 6f 20 77 6f 72 6c 64 0a 0b 0b 1b 64 02 1b 40>
PS D:\development\node\testposprinter> 

``` 

But no ticket !
[image]

jayel74, 2023-10-15:
hello,

No update on the above?

Klemen1337, 2025-05-28:
Did you ever get it to work?

Closing this issue because it is stale