Klemen1337/node-thermal-printer #238: Will it work for non popular brands of printer

unaisshemim, 2023-10-05:
i am using POS-80-Series brand this name is getted from the printer itself and it is connected and make print through chrome .i want make sure that will it work through this **node thermal printer also**

nipoonp, 2023-10-08:
I got some cheap Chinese printers and they work. I think if the printers see a Epson print simulation it will work?

mderler, 2023-10-10:
I used the library on a Citizen printer and with the epson configuration worked. But maybe you can find a sheet of the byte codes for each command your printer uses. If your printer uses an other set of byte codes you probably have to write your own config file. Take a look into /lib/types

krtschmr, 2023-11-08:
> I got some cheap Chinese printers and they work. I think if the printers see a Epson print simulation it will work?

https://www.amazon.de/gp/product/B07QWSDP2K/ref=ppx_yo_dt_b_asin_title_o00_s00?ie=UTF8&th=1

like those?MUNBYN Thermal Printer

Abe-Telo, 2023-12-13:
I am planning on getting the CITIZEN CL-E300 DT PRINTER, i hope it will work.

skylarmt, 2024-01-12:
I've messed around with lots of sketchy receipt printers. Most cheap Chinese receipt printers speak ESC/POS (Epson) out of the box and are perfectly happy being configured as an Epson thermal printer in whatever software you use.  Generally, if you can open them as a serial port, send a line of text, and the text prints, it's all good.  On Linux, just `echo "Hello World" > /dev/usb/lp0`. Substitute the path for whatever your printer is, some show up as a plain serial port like `/dev/ttyACM0` or `/dev/ttyUSB0`.

As for @unaisshemim's POS-80 printers, IIRC the first time I tried to use one (many years ago now, circa 2012) it needed a special Windows-only driver. I contacted the manufacturer and miraculously they sent me a special firmware file that made the printer appear as a USB serial adapter instead, so I could send it ESC/POS commands over serial. I still have one in use today, but that helpful firmware file is long gone (and so is the company I worked for, I got their cash registers during their liquidation).

Hope this bit of general receipt printer knowledge helps someone.  tl;dr yes your Aliexpress receipt printer will work, it's likely an Epson clone.

adnanlah, 2024-03-30:
I tried it with the X-Printer Q371U/T371U printer and it worked very well.