Klemen1337/node-thermal-printer #239: Data Sent To Printer - hang

ggb88, 2023-10-19:
Hi all.  Was using 4.2.1 and all good.  Upgraded to latest version 4.4.x and when performing print, I get the following output on the console

```
unity_print 2023-10-20T10:48:12.752+11:00 [debug]:      Printer connected at Epping 
Data sent to printer: 10.20.0.88:9100 <Buffer 1b 74 00 1d 42 00 1b 61 01 1d 21 33 41 50 43 4f 20 41 50 50 0a 1d 42 01 1b 61 01 1d 21 11 49 4e 20 53 54 4f 52 45 0a 0a 1d 21 00 1b 21 00 1d 42 00 1b ... 265 more bytes>
```

and the process hangs.  Maybe I've missed something in the updates and now there is a requirement to do something else.  I've reverted back for the time being as I have a release due ASAP.  if I get the chance I'll investigate further.

pierrepaz, 2023-10-20:
Same here, with Epson TM-T20X Ethernet, just if using promisse, for sure.

ggb88, 2023-10-20:
We are using epson TM88-VI - ethernet

thomaslc08, 2023-10-21:
Same with TM 30 II

Klemen1337, 2023-11-06:
The code bellow works fine for me. Please provice your code examples.

```node
const { ThermalPrinter, PrinterTypes, CharacterSet, BreakLine } = require('node-thermal-printer')

async function example() {
  const printer = new ThermalPrinter({
    type: PrinterTypes.EPSON,
    interface: "tcp://192.168.1.87:9100",
    options: {
      timeout: 1000,
    },
    width: 48,
    characterSet: CharacterSet.SLOVENIA,
    breakLine: BreakLine.WORD,
    removeSpecialCharacters: false,
    lineCharacter: '-',
  });

  printer.println('Hello World!');
  printer.cut();

  try {
    await printer.execute();
    console.log('Print success.');
  } catch (error) {
    console.error('Print error:', error);
  }
}

example();
```

vraristizabal, 2023-11-12:
I think the issue is in the Network interface, it may be missing to resolve the promise?

```
networkConnection.write(buffer, null, () => {
            console.log(`Data sent to printer: ${name}`, buffer);
            if (!options.waitForResponse) {
              networkConnection.destroy();
              resolve();
            }
          });
```

treeindark, 2023-12-22:
did you plan to add the resolve ? (it work with it)

Klemen1337, 2024-01-23:
The pull request was accepted and it will be applied in the new version

closed 2024-01-23 by Klemen1337