Klemen1337/node-thermal-printer #245: Cyrillic printing. setTypeFontB

fobus821, 2024-01-16:
I use the GOOJPRT MPT-3 printer
Firmware: **MTP-3B-YC-8008**
Font: **1.10**
The problem is that if you use setTypeFontB, the Cyrillic alphabet is printed incorrectly. If you use setTypeFontA or setTextNormal, everything is printed normally. Note: On firmware MTP-3B-YC-6107 and Font: 1.09, everything works fine. Is there a solution in the code? Is it necessary to flash to an older firmware?

olimexsmart, 2024-02-26:
This seems more a problem with the specific printer firmware

But still, have you tried setting Cyrillic when setting up the printer?
```
  let printer = new ThermalPrinter({
      type: PrinterTypes.EPSON,
      interface: `tcp://${pInfo.ip}:${pInfo.port}`,
      characterSet: CharacterSet.PC855_CYRILLIC, // or PC855_CYRILLIC2
  });
```

Klemen1337, 2025-05-28:
Closing because it is stale