Klemen1337/node-thermal-printer #248: How does printing labels work?

adnanlah, 2024-03-30:
Hello,

I tried this packages for printing receipts and it worked pretty well. However I am yet to try it with a label only printer so now I am wondering how printing labels work, I mean how do developer use this package to make the printer jump to the next label with no gaps or shifting?

I hope you understood my question. I would really appreciate any ideas on this.

Thank you very much everyone and especially whoever contributed to this work.

OverSamu, 2024-03-30:
Hi, try looking at this issue/comment: https://github.com/Klemen1337/node-thermal-printer/issues/150#issuecomment-663379240

adnanlah, 2024-03-30:
> Hi, try looking at this issue/comment: https://github.com/Klemen1337/node-thermal-printer/issues/150#issuecomment-663379240

Thank you very much. 

I have tried out the x-printer Q371U that supports both receipt model and label model. This package and other packages work fine when the printer is in the receipt mode, but the printer doesn't print or even scroll if it's in the label mode.

I thought the problem was from the printer label mode is broken, now I'm starting to think it's not.

I still need to try out other label printers though.

adnanlah, 2024-04-02:
I learned that label printers support programming languages like TSPL, ZPL, EPL. Here is a code example on how to use TSPL with `node-thermal-printer` that could provide utility to someone:

```ts
const cmds = [
  'SIZE 48 mm,25 mm',
  'CLS',
  'TEXT 10,10,"4",0,1,1,"Hello world!"',
  'BARCODE 10,60,"128",90,1,0,2,2,"012345678901234"',
  'PRINT 1',
  'END',
];

printer.raw(Buffer.from(cmds.join('\r\n'))
```

Here is a nice tutorial about this subject: https://hackernoon.com/how-to-print-labels-with-tspl-and-javascript

closed 2024-04-02 by adnanlah