Klemen1337/node-thermal-printer #250: Is it possible to control the size, margins, and position of an image?

hugocalheira, 2024-04-04:
I need to upscale an image that I'm printing. How do I do this?

```import { ThermalPrinter, PrinterTypes, CharacterSet, BreakLine } from 'node-thermal-printer';
import driver from 'printer'
import axios from 'axios'

const printersList = {
  TMT88V: 'printer:EPSON TM-T88V Receipt'
}

let printer = new ThermalPrinter({
  type: PrinterTypes.EPSON,
  interface: printersList.TMT88V,
  driver,
  characterSet: CharacterSet.PC860_PORTUGUESE,
  removeSpecialCharacters: false,
  breakLine: BreakLine.NONE,
  options:{
    timeout: 5000
  },
  width: 42
});

// this function returns me the image that will be printed
async function getLayout() {
  const url = 'URL'
  const payload = {"canal":1,"cdFilial":1,"cdOperador":0,"idCliente":1}
  const result = await axios.post(url, payload, { responseType: 'arraybuffer' })
  await printFromBuffer(result.data)
}

async function printFromBuffer(buffer) {
  try {
    let isConnected = await printer.isPrinterConnected()
    if (!isConnected) {
      throw Error("not connected")
    }
    
    await printer.printImageBuffer(buffer)
    printer.cut()
    await printer.execute()
    printer.clear()
  } catch (e) {
    console.log(e)
  }
}

getLayout()
```

The image size is 450x1098px and it prints almost perfectly, but it doesn't occupy all the area on the paper.
[image]

adnanlah, 2024-04-06:
I don't think you can with this package.

You can check out the issue I posted few days ago https://github.com/Klemen1337/node-thermal-printer/issues/249. In my case I had to print images in `580px` width in order of making the image to take full width of the paper (`80mm`) on my printer as I didn't try other printers yet.

Try to print images that has `580px` width and see if it's common across printers or not. If it worked then you have to use some node.js package to upscale/downscale images to `580px` width before printing it.

I would be happy if you let me know how did it go.

Good luck.

hugocalheira, 2024-04-08:
I solved it in a similar way. I upscaled by intercepting the image buffer, changing the width, and generating a new buffer for printing. That way, I got the result I needed in this case. The issue is that depending on the desired size, the final printing speed is significantly impacted, making the process slower.

But I miss having a better way to handle printing, including positioning, scaling, or even controlling print margins.

adnanlah, 2024-04-08:
> changing the width

Good job you made it work. Would you please help me with the image width that was enough to make the image occupy all width of the paper?

closed 2025-05-28 by Klemen1337