Klemen1337/node-thermal-printer #268: Unable to get printer interface

amrutakawale, 2025-03-03:
I have implemented this library for the STAR TSP100III printer, but code is not working as I am using printer which is connected by USB.
Code is broken at this line: **interface: `tcp://${printerInfo.name}`,  // Printer interface (assuming network interface here) **

ipcMain.on("printOrderInfo", (event, count, printContent, isCashOrCheck) => {

  // Get available printers

  const printers = mainWindow.webContents.getPrinters();

  // Find the printer named "Star TSP100 Cutter"

  let printerInfo = printers.find((printer) =>

    printer.name.includes("Star TSP100 Cutter")

  );
 
  // Check if the printer is available

  if (!printerInfo) {

    console.log("Star TSP100 Cutter Printer is not available for Print.");

    return;

  }
 
  // Initialize the printer instance

  const printer = new ThermalPrinter({

    type: PrinterTypes.STAR,  // Printer type: 'star' or 'epson'

    interface: `tcp://${printerInfo.name}`,  // Printer interface (assuming network interface here) 

    characterSet: CharacterSet.PC852_LATIN2,  // Printer character set

    removeSpecialCharacters: false,  // Do not remove special characters by default

    lineCharacter: "=",  // Character for lines in the receipt (default: "-")

    breakLine: BreakLine.WORD,  // Break line after word (default: WORD)

    options: {

      preview: false,  // No preview

      width: "240px",  // Set content body width

      margin: "0 0 0 0",  // Set margin of content body

      copies: 1,  // Number of copies to print

      printerName: printerInfo.name,  // Printer name from the available printers list

      timeOutPerLine: 400,  // Timeout per line for printing

      silent: true,  // Do not show preview or print dialogs

      timeout: 5000  // Timeout for connection (network printers)

    }

  });
 
  // Loop through each print request and print

  for (let index = 0; index < count; index++) {

    const data = [

      {

        type: "text",  // Type of data: text

        value: printContent[index],  // Text content to print

        css: {

          "text-align": "center",  // Center align the text

        },

      }

    ];
 
    // Add the raw command to open the cash drawer if `isCashOrCheck` is true

    if (isCashOrCheck) {

      // Raw command for cash drawer (e.g., BEL \x07)

      data.push({

        type: "raw",  // Use 'raw' type for sending control commands

        value: "\x07",  // ASCII BEL (bell) to open cash drawer

      });

    }
 
    try {

      // Print the text data

      await printer.printText(data);  // Print the content as text

      await printer.cut();  // Cut the paper after printing
 
      // If the transaction is cash or check, print the raw command to open the drawer

      if (isCashOrCheck) {

        const rawCommand = Buffer.from([0x07]);  // Convert \x07 to a buffer for raw printing

        await printer.printRaw(rawCommand);  // Send the raw command

      }

    } catch (error) {

      // Handle printing errors

      console.error("Failed to print. Error: " + error);

    }

  }

});

Please help me to resolve it.
Thank you in advance!

yashkolambekar, 2025-03-28:
it will be different for windows, it will be start with `COM`.  you may find it in your device manager or something like that.

amrutakawale, 2025-04-01:
As the printer is connected through USB, it can't use the COM port name. Could you please suggest anything else?

Thank you!

yashkolambekar, 2025-04-01:
Are you sure? Because when I used to connect my Arduino to windows laptop, it was mounted at COM7 or something like that, can you verify once?

amrutakawale, 2025-04-01:
Yes I already verified and it is showing USB001 for printer

yashkolambekar, 2025-04-01:
@amrutakawale can you please send your repo on my mail, I will try to reproduce it.

alias@yassh.in

amrutakawale, 2025-04-02:
@yashkolambekar I sent you the code details. Could you please check once and let me know if you require anything else from my side?

Thank you!

amrutakawale, 2025-06-04:
> Did you solve the issue?

No I am not able to solve this issue.
Thanks & Regards,
Amruta Kawale
9850157194